fbiego/esp32-c3-mini #7: Support for TTGO Watches

alpha815, 2024-06-22:
Hi,

Thank you for this amazing project! I truly appreciate all the hard work and dedication put into it.

I would like to inquire if there are any plans to support the TTGO Watch product lines. They have some excellent devices in their lineup, that are also well supported within the Arduino framework. It would be wonderful to see these devices integrated with your project.

Thank you for your time and consideration.
[https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library](url)

fbiego, 2024-06-22:
Thank you.
Unfortunately, I do not have any TTGO boards as of now therefore I'm not able to test and add support

alpha815, 2024-06-26:
I might be able to add support for the TTGO T-Watch v2 as I have this version available. I will start by getting the user interface working. TTGO watches require initial configuration due to their Power Management Unit (PMU), but that should not be an issue. After that, do I need to configure the IMU for step counting? Is this feature available in the current firmware? Could you provide additional instructions?

fbiego, 2024-06-26:
The project does not include any external sensors/peripherals interface. It is only the UI interface based on LVGL. This is because the original C3-Mini board does not have any IMU. This can however be included based on the board. 
You can fork the repo, add the configurations then test the board.