fbiego/esp32-c3-mini #9: Dram reduction possibility

prashantchawla, 2024-08-10:
Hi Fbiego,

Great work - you have taken ESP32 development to next level. Thank you for making your work open source.

I was trying to make the project work for ESP32 Wroom display module. The compilation goes fine, but I get an error that the dram segment is overflowing by 49600 bytes.

I have tried to reduce the buffer size in a few places, but haven't been able to get any substantial reduction. Any direction to resolve this that you would be able to give will be appreciated.

fbiego, 2024-08-12:
try reducing the lvgl buffer size, this will affect performance 

https://github.com/fbiego/esp32-c3-mini/blob/b5a1ec714109f68eb37ba48952cf67719f049bee/include/lv_conf.h#L53

try `70U*1024U`

prashantchawla, 2024-08-13:
Thank you, I was able to successfully compile and upload the build now. Had to reduce till `59U*1024U` as was also messing around with some other buffers/caches in parallel. 

The build starts fine and I can see the screen with heading 'ESP32 C3 Mini' and 4 icons. However, it just freezes there. Enabled logging for LVGL and found that it is getting out of memory error. Last few logs:

`[Info]  (3.962, +11)     lv_qrcode_create: begin        (in lv_qrcode.c line #61)
[Info]  (3.987, +25)     lv_label_create: begin         (in lv_label.c line #75)
[Info]  (3.990, +3)      lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (3.991, +1)      lv_img_create: begin   (in lv_img.c line #60)
[Info]  (4.003, +12)     lv_qrcode_create: begin        (in lv_qrcode.c line #61)
[Info]  (4.021, +18)     lv_label_create: begin         (in lv_label.c line #75)
[Info]  (4.023, +2)      lv_obj_create: begin   (in lv_obj.c line #215)
[Info]  (4.024, +1)      lv_img_create: begin   (in lv_img.c line #60)
[Info]  (4.036, +12)     lv_qrcode_create: begin        (in lv_qrcode.c line #61)
[Info]  (4.036, +0)      lv_mem_alloc: couldn't allocate memory (2858 bytes)    (in lv_mem.c line #140)
[Info]  (4.047, +11)     lv_mem_alloc: used:  58292 ( 97 %), frag:   0 %, biggest free:   2124  (in lv_mem.c line #146)
[Error] (4.057, +10)     lv_qrcode_constructor: Asserted at expression: buf != NULL (Out of memory)     (in lv_qrcode.c line #196)`

Any idea which buffer reduction would have caused this or how to proceed? Or taking a step back, if running with such constrained resources is even possible? 

Again, thank you for the project and your help.

fbiego, 2024-08-15:
You might want to check the latest push, the board config below disables custom watchface support and lowers mem size

https://github.com/fbiego/esp32-c3-mini/blob/dfc427e60226e6ef6abf7929c7cf4630f28fddca/platformio.ini#L167

prashantchawla, 2024-08-26:
I am able to use the project fine on the screen now - but as you pointed out, performance is not the best.
Thanks @fbiego for your help.

closed 2025-04-05 by fbiego