fbiego/esp32-c3-mini #12: Cant recieve notification message from bluetoothctl

AndhikaFW, 2024-09-06:
Im trying to get my custom message notification client working for my Ai project. I've managed to send data through the uart gatt attribute, I know this because my esp32 crashed when I overload the data. However there's no notification appeared on my esp32. I've read the ChronosESP32.h, is there anything I might be missing? How can I able to implement custom notification client for esp32-c3-mini?

fbiego, 2024-09-06:
Can you provide more context on what you are trying to achieve?
How and where are you sending the data from?

AndhikaFW, 2024-09-06:
> Can you provide more context on what you are trying to achieve? How and where are you sending the data from?

Well I'm developing an Ai assistant program for raspberry pi. I use esp32 for displaying Ai responses and interacting with the Ai. I'm trying to do gatt client data transfer to esp32 with bluetoothctl to better understand how I can write the bluetooth interface for my Ai program.

I was sending the data from fedora linux.
I used bluetoothctl as the program.
I selected the UART attribute to do data transfer into (the one with write-without-response characteristic).
I transferred "0xAB 0x00 0x00 0x00 0x72 0x00 0x03 0x02 0x01", which I assume would be processed as notification message after reading the chronosESP32.cpp code.
It then proceed with success on bluetoothctl but nothing happens on esp32 side.

AndhikaFW, 2024-09-06:
> Can you provide more context on what you are trying to achieve? How and where are you sending the data from?

I also have tried the nrf scanned app on android

fbiego, 2024-09-06:
Got it. I think what you are looking for is the protocol. The command you used above might not be correct.
Currently, everything is managed by the Chronos app. But as for the protocol, you can refer here
https://github.com/fbiego/dt78/blob/master/DT78_COMMANDS.md#messages
I will try to setup a python script with the protocol for sending messages to test the same

AndhikaFW, 2024-09-06:
> Got it. I think what you are looking for is the protocol. The command you used above might not be correct. Currently, everything is managed by the Chronos app. But as for the protocol, you can refer [...]

Ah thanks. Im closing this issue now.