fbiego/esp32-c3-mini #15: First Compile and run issue

itspoppadom, 2024-12-16:
Hey there, brilliant project! Excited to run this on my hardware, but i've encountered some issues with compiling and running the code. 

Using platform.io to compile and run the project after selecting the correct board in platform.ini file. 

I have encountered following issues during compile relating to iphone bluetooth  ble advertisement

Any ideas on how to resolve? Instinct was tellign me to just remove these and hope for the best since I'm using android there shouldnt be issues

Archiving .pio\build\lolin_s3_mini_1_69\libb8c\libPreferences.a
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\FirmwareMSC.cpp.o
.pio/libdeps/lolin_s3_mini_1_69/ChronosESP32/src/ChronosESP32.cpp: In member function 'void ChronosESP32::begin()':
.pio/libdeps/lolin_s3_mini_1_69/ChronosESP32/src/ChronosESP32.cpp:112:16: error: 'class NimBLEAdvertising' has no member named 'setScanResponse'; did you mean 'setScanResponseData'?
  pAdvertising->setScanResponse(true);
                ^~~~~~~~~~~~~~~
                setScanResponseData
.pio/libdeps/lolin_s3_mini_1_69/ChronosESP32/src/ChronosESP32.cpp:113:16: error: 'class NimBLEAdvertising' has no member named 'setMinPreferred'; did you mean 'setMinInterval'?
  pAdvertising->setMinPreferred(0x06); // functions that help with iPhone connections issue
                ^~~~~~~~~~~~~~~
                setMinInterval
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\FunctionalInterrupt.cpp.o
.pio/libdeps/lolin_s3_mini_1_69/ChronosESP32/src/ChronosESP32.cpp:114:16: error: 'class NimBLEAdvertising' has no member named 'setMinPreferred'; did you mean 'setMinInterval'?
  pAdvertising->setMinPreferred(0x12);
                ^~~~~~~~~~~~~~~
                setMinInterval
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\HWCDC.cpp.o
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\IPv6Address.cpp.o
Compiling .pio\build\lolin_s3_mini_1_69\FrameworkArduino\MD5Builder.cpp.o
*** [.pio\build\lolin_s3_mini_1_69\lib17c\ChronosESP32\ChronosESP32.cpp.o] Error 1

fbiego, 2024-12-16:
Are you using v2 of the NimBLE library? There has been changes in the API. Currently, v1.4 of NimBLE is used by ChronosESP32 library

itspoppadom, 2024-12-16:
After some struggle, it appears that changing to 1.4.3 seems to have fixed the issue at compile, but produced new issues at upload using the esptool? 

Compressed 4054752 bytes to 1020370...
Writing at 0x00010000... (1 %)
Writing at 0x0005b2a8... (3 %)
Writing at 0x000a606e... (4 %)
Writing at 0x000cef84... (6 %)
Writing at 0x00101698... (7 %)
Writing at 0x0011dc8e... (9 %)
Writing at 0x0013b6e1... (11 %)
Writing at 0x001490ca... (12 %)
Writing at 0x0015b03e... (14 %)
Writing at 0x00162c81... (15 %)
Writing at 0x00189745... (17 %)
Writing at 0x001c20d5... (19 %)
Writing at 0x001d0c8e... (20 %)
Writing at 0x001ecda4... (22 %)
Writing at 0x0020190e... (23 %)
Writing at 0x00228cef... (25 %)
Writing at 0x0025cab8... (26 %)
Writing at 0x0028743d... (28 %)
Writing at 0x002a9614... (30 %)
Writing at 0x002b5894... (31 %)
Writing at 0x002be060... (33 %)
Writing at 0x002c6cb1... (34 %)
Writing at 0x002d1cfe... (36 %)
Writing at 0x002df12a... (38 %)
Writing at 0x002eee13... (39 %)
Writing at 0x00302688... (41 %)
Writing at 0x0030d9f5... (42 %)
Traceback (most recent call last):
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 1032, in _main
    main()
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 832, in main
    operation_func(esp, args)
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\cmds.py", line 586, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 131, in inner
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 1030, in flash_defl_block
    self.check_command(
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 404, in command
    p = self.read()
        ^^^^^^^^^^^
  File "C:\Users\domin\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 337, in read
    return next(self._slip_reader)
           ^^^^^^^^^^^^^^^^^^^^^^^
StopIteration

A fatal error occurred: The chip stopped responding.
*** [upload] Error 2

itspoppadom, 2024-12-16:
Ah the latest appeared to be an issue on my side, re flashed the ESP32 with examples from ChronosESP32 Library in Arduino IDE to ensure that there are no issues with the board. tried to re upload again using Platform.io and the current repo. 
And finally Success... Using the Waveshare ESP32-S3-Touch-LCD-1.69 

Pretty much works straight out the box

jlusco, 2024-12-16:
@itspoppadom Do you get reboots with the waveshare

itspoppadom, 2024-12-16:
@jlusco I did not experience any un assisted reboots. Only reboots experienced were intentional through the use of the reset button on the board

closed 2025-04-05 by fbiego