fbiego/esp32-c3-mini #21: Support for seeedstudio esp32s3 and xiao display

vip20, 2025-04-10:
Does this work for seeedstudio xiao devices

fbiego, 2025-04-10:
Can you give specifications or links to the boards?
Typically, if it has similar screens, then its a matter of changing the pin assignments

vip20, 2025-04-11:
Board: https://www.seeedstudio.com/Seeed-Studio-XIAO-ESP32S3-Pre-Soldered-p-6334.html
Board Datasheer: https://files.seeedstudio.com/wiki/SeeedStudio-XIAO-ESP32S3/res/esp32-s3_datasheet.pdf

Display: https://www.seeedstudio.com/Seeed-Studio-Round-Display-for-XIAO-p-5638.html

I added this in pins

```

#elif XIAO_ESP32S3

// screen configs
#define WIDTH 240
#define HEIGHT 240
#define OFFSET_X 0
#define OFFSET_Y 0
#define RGB_ORDER false

// touch
#define I2C_SDA 3
#define I2C_SCL 2
#define TP_INT 21
#define TP_RST -1

// display
#define SPI SPI2_HOST

#define SCLK 10
#define MOSI 9
#define MISO 8
#define DC 4
#define CS 2
#define RST 13

#define BL 43

#define BUZZER -1

#define MAX_FILE_OPEN 50
```

but I got "Setup FS failed" error in Serial Monitor and the display is blank.

The display has a sd card support. I was able to use it in my separate project

```
void initializeSD()
{
    // Initialize SD card
    int attempts = 0;
    int maxAttempts = 50;
    int delayBetweenAttempts = 300;

    pinMode(D2, OUTPUT);
    while (!SD.begin(D2))
    {
        Serial.print("Attempt::::");
        Serial.println(attempts);
        // Serial.println("SD Card mount failed! (attempt %d of %d)", attempts, maxAttempts);
        attempts++;
        if (attempts > maxAttempts)
        {
            Serial.println("Giving up");
        }
        delay(delayBetweenAttempts);
    }

    Serial.println("SD Connected");
}
```

fbiego, 2025-04-11:
does the screen work without SD?

fbiego, 2025-04-11:
i think you need to set the spi bus shared to true
https://github.com/fbiego/esp32-c3-mini/blob/7955dc19059ec0b131871f5e2aa46346521cd02b/hal/esp32/displays/generic.hpp#L63

vip20, 2025-04-11:
I shall check post this weekend and keep you posted. Thanks for the quick reply

vip20, 2025-04-15:
@fbiego , I tried this and it is not working

fbiego, 2025-04-16:
> The display has a sd card support. I was able to use it in my separate project

What library did you use for the display in your project?

vip20, 2025-04-16:
I'm using lvgl for display. Also I followed seeedstudio display

fbiego, 2025-04-17:
This project uses [LovyanGFX](https://github.com/lovyan03/LovyanGFX) for the display drivers. I currently don't have a similar board with SD card support, so I can't test that functionality at the moment

You'll likely need to modify the code and ensure you're using the same driver to enable SD card support.