fbiego/esp32-c3-mini #24: (HELP!) collect2.exe: error while building for windows_64

pradeethd, 2025-04-14:
Hello Sir, me again. Thanks for this wonderful project and helping me. I have one another issue with building the project for emulating in my PC. Pls help me with this too. Thank you.
 
Linking .pio\build\windows_64\program.exe
D:/Program Files/MSYS2/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/14.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lD:/Works/Projects/SusGameWatch/esp32-c3-mini-main/Files/MSYS2/mingw64/lib: Invalid argument
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\windows_64\program.exe] Error 1

fbiego, 2025-04-14:
have you set the lines below according to your path?
https://github.com/fbiego/esp32-c3-mini/blob/6eb8a61e9d53d0f0fe0ea662bbfb25fb07011dbb/platformio.ini#L67-L68

pradeethd, 2025-04-14:
Yes. D:/Program Files/MSYS2/mingw64/bin and D:/Program Files/MSYS2/mingw64/include/SDL2/SDL.h. These are my directories. It is working fine with the lvgl demo project link you gave in the ReadMe page.

pradeethd, 2025-04-14:
Sorry not fixed yet

fbiego, 2025-04-14:
> D:/Program Files/MSYS2/mingw64/bin

It should be `lib` folder not `bin`

```
-L D:/Program Files/MSYS2/mingw64/lib/
-D SDL_INCLUDE_PATH="\"D:/Program Files/MSYS2/mingw64/include/SDL2/SDL.h\"" 
```

pradeethd, 2025-04-14:
It was lib only in the code. I mistyped.

fbiego, 2025-04-14:
try commenting it out
`; -L D:/Program Files/MSYS2/mingw64/lib/ ; comment or remove`

pradeethd, 2025-04-14:
Now it's working. Thank you so much.

closed 2025-04-14 by pradeethd