fbiego/esp32-c3-mini #30: Support for Waveshare ESP32-S3 1.43inch AMOLED Display

cddmp, 2025-06-16:
Hi,

first of all thank you for all your work you put into this. Looks great to me!
Withoug digging too much into the code, how difficult would it be to add this display:
https://www.waveshare.com/esp32-s3-touch-amoled-1.43.htm

fbiego, 2025-06-21:
seems the configuration is almost similar to the Viewe boards
https://github.com/fbiego/esp32-c3-mini/blob/07e8b48f353db3f37ad4a3cfb202aab5a4af1682/platformio.ini#L18

Can you try it?
You may need to change pinout
https://github.com/fbiego/esp32-c3-mini/blob/07e8b48f353db3f37ad4a3cfb202aab5a4af1682/hal/esp32/displays/pins.h#L169-L184

cddmp, 2025-06-22:
I will have a look at the touch driver.

kolioDev, 2025-10-25:
> I have a working proof of concept. I will report back.

Hi, can you share how you got the touch screen working. I am working with a board with CST9217 touch driver (Waveshare ESP32-S3-Touch-LCD-1.28) and was curious to see how I can integrate the touch screen.