fbiego/esp32-c3-mini #32: ESP32-S3 (WaveShare S3 with 1.28) not working

smashingchrische, 2025-06-28:
Hello,
I have build the firmware for the waveshare lcd board:
https://www.waveshare.com/product/esp32-s3-touch-lcd-1.28.htm

The build works fine (except some compiler warnings) and flashes the device, however the esp32-s3-1.28 displays stays dark.

The board should be fine (the original demo application was working).
I have tried to build previous versions by checking out elder commits, but the build process seems to fail there.
I will try to fix if I have some time, but maybe someone from the team will see the issue directly.

Regards,
Christian

smashingchrische, 2025-06-28:
The esp32 boots up, but the screen has a timeout and the filesystem setup fails:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x19 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x41c
load:0x403c9700,len:0x9a8
load:0x403cc700,len:0x28ac
entry 0x403c98b8
[INFO]: Starting up device
Setup FS failed
Drive S is ready
Total: 308396   Free: 138480    55.10%
[INFO]: Setup done
[INFO]: v6.0 [fbiego]
ESP32-S3
42:aa:bb:cc:bb:aa
[WARNING]: Screen timeout

fbiego, 2025-06-30:
You can make the changes as below;

https://github.com/fbiego/esp32-c3-mini/blob/07e8b48f353db3f37ad4a3cfb202aab5a4af1682/hal/esp32/app_hal.cpp#L1904
```
int tm = prefs.getInt("timeout", 4); // Set the timeout to always on
```

https://github.com/fbiego/esp32-c3-mini/blob/07e8b48f353db3f37ad4a3cfb202aab5a4af1682/hal/esp32/app_hal.cpp#L1909
```
navSwitch = prefs.getBool("autonav", true); //Navigation will begin automatically when started from the phone
```